bxparks/EpoxyDuino #2: piping to a screen pager (e.g. less(1) or more(1)) blocks program

bxparks, 2019-08-13:
As noted in the README.md, if the executable (e.g. `SampleTest.out`) is piped to a screen pager like `less(1)` or `more(1)`, sometimes (not all the time) the executable hangs and displays nothing on the pager program. The program seems to be blocking on the standard input. The problem is probably caused by an interaction between how the screen pager manipulates the `stdin` and how I place the stdin into "raw" mode in `Arduino.cpp`. I would love to fix this, but I don't know enough about Unix tty device configuration to figure this out.

In the meantime, the workaround is to explicitly redirect the `stdin` using `/dev/null`, like this:
```
$ ./SampleTest.out | less # hangs
$ ./SampleTest.out < /dev/null | less # works
```

mmrazik, 2021-03-20:
@bxparks please see my pull request

bxparks, 2021-03-22:
@mmrazik: Let's transfer the discussion from https://github.com/bxparks/EpoxyDuino/pull/24 to here. 

The number of lines needed to trigger the error seems to be variable. Changes across machine and OS, sometimes different on the same machine. I checked in a version of `PipeFail` that prints out 2000 lines, and that seems to trigger this problem on all machines that I tested today:

* Ubuntu MATE 20.04
* Linux Mint 20 (20.04 based)
* Mac OS 10.13.3 (Mojave)
* FreeBSD 12

bxparks, 2021-09-30:
I believe I have finally fixed this with bf6248f.

closed 2021-09-30 by bxparks