bxparks/EpoxyDuino #7: Isue with Makefile

synergiator, 2020-03-21:
Hi there,

any many greetings from the Open Source Ventilator project!

I have tried to integrate available code with UnixHostDuino but get this error:
`
make: *** No rule to make target 'OpenSoftwareVentilator.ino', needed by 'OpenSoftwareVentilator.o'.  Stop.`

Code:
https://github.com/covid-supplies/Open-Source-Ventilator/blob/master/Dockerfile

bxparks, 2020-03-21:
Hi,

The first problem is that your Makefile contains
```
APP_NAME := OpenSoftwareVentilato
```
instead of
```
APP_NAME := OpenSourceVentilator
```

But even if you fix that typo, UnixHostDuino will not work for your project, because UnixHostDuino is not meant to support all (many?) Arduino programs. There are a lot of hardware dependent code that won't compile on a Unix machine. One of the first things that I hit was `SPI.h`, which pulls in hardware dependent parameters like `F_CPU`, `SPIF`, `SPE`, MSTR`, `SREG`, `SPSR`, `noInterrupts()`, and so on. I don't even know what most of those parameters are.

If you are looking to build for continuous integration, you might find one of my other projects helpful, https://github.com/bxparks/AUniter, but I'm not sure if that will meet your needs.

bxparks, 2020-03-27:
I'm going to close this if you have no further questions.

closed 2020-03-27 by bxparks