bxparks/EpoxyDuino #11: Arduino CLI integration

Legion2, 2020-06-08:
The Arduino build system is very modular you can add your own libraries, boards and build tools.
Would it be possible to make UnixHostDuino a build tool which can be used from ArduinoIDE and ArduinoCLI. For example UnixHostDuino could provide an own [platform](https://arduino.github.io/arduino-cli/platform-specification/) with the mocks and build tools. In the ArduinoIDE we the select "UnixHostDuino" instead of "Arduino Uno" as board and then can "verify" the sketch, which compiles the sketch to a native executable and place it in the sketch folder.

https://arduino.github.io/arduino-cli/sketch-build-process/

bxparks, 2020-06-09:
Umm, one of the primary goals of UnixHostDuino was to be completely free of the Arduino IDE tool chain because it is so cumbersome. This allows UnixHostDuino to be used for continuous integration using a simple Docker image, for example. Keep in mind that UnixHostDuino provides only rudimentary Arduino emulation. It will never compile most Arduino programs. What is your use case?

Legion2, 2020-06-09:
Arduino CLI is decoupled from ArduinoIDE it's the new Backend of the eco system. Arduino CLI can be used in CI pipelines and there is an official GitHub Action to run it in the GitHub Actions CI.
https://github.com/arduino/arduino-cli

bxparks, 2020-06-09:
The Arduino IDE and CLI produce binaries that run on the target microcontroller. The purpose of UnixHostDuino is to produce binaries that can run on the Unix host. If you are suggesting that I can use Arduino CLI to replace a simple `make` command, perhaps. But that's a heck a of lot work, additional maintenance overhead, and extra layers of abstractions in the build pipeline. I'm not seeing any compelling benefits of that work, especially for something that won't compile most Arduino programs. So I'm still asking, what is your use case?

Legion2, 2020-06-09:
I have this arduino library https://github.com/Legion2/CorsairLightingProtocol and I want to do unit test for code that does not interact with the hardware. The Unit tests must run in CI pipeline and should be portable which mean they must be also run on local machine. The installation of the tools should be straightforward and no special setup should be required. The library must be useable by people only using Arduino IDE, the unit test should not interfere with the existing setup.

bxparks, 2020-06-10:
The Arduino CLI is separated from the IDE, but it still requires the compiler tool chain for the various target microcontrollers (e.g. avr-gcc, xtensa-lx106-elf-gcc, xtensa-esp32-elf-gcc), for each of the supported OS's (MacOS, Linux, Windows). Configuring something similar for a Unix target, figuring out if it's even possible to configure the boards.txt, platform.txt, etc. to work on a Unix machine, is far beyond what I want to spend time on.

I generally assume that the users of my Arduino libraries are not interested in running the unit tests. So those unit tests are written for my own benefit. The process that seems to work for me is that I start writing the AUnit unit test on the Arduino IDE. The test will run on microcontroller itself. After writing a lot of tests, I become tired of running those tests manually through the IDE. I suppose I could use the Arduino CLI, but I don't use the CLI because when I looked at a while ago, it wasn't functional enough. (I haven't had the time to look at it more recently).  Then I try to make the AUnit test run under UnixHostDuino by creating a Makefile in the same AUnit test folder. Usually, a given AUnit test will run under **both** Arduino IDE and UnixHostDuino. Occasionally, I write unit tests which are so large, they will run *only* under UnixHostDuino.

I don't know if that helped, but the short answer is that UnixHostDuino will probably never support that usage that you initially asked for.

bxparks, 2020-06-10:
If you want to see an example of how this works, take a look at:
https://github.com/bxparks/AceTime/tree/develop/tests
I think every one of those unit tests works under both Arduino IDE and UnixHostDuino.

bxparks, 2020-06-23:
I'm going to close this. The summary is that UnixHostDuino was created for the explicit purpose of avoiding the Arduino tool chain, including the IDE and CLI, so there is no compelling reason to spend my time and energy on this. But if you figure out how to make it work, more power to you.

closed 2020-06-23 by bxparks