bxparks/EpoxyDuino #20: Using avr functions in EEPROM.h

lkankowski, 2021-02-18:
I cant compile EEPROM.h on my laptop because the are used avr functions:
- eeprom_read_byte
- eeprom_write_byte
Is that correct?

I am using Platformio, not makefile, but I think it is not correct.

bxparks, 2021-02-18:
You are correct, EEPROM.h is not functional. When I committed that version, I made the smallest change possible to allow some of my programs compile. But if you actually try to use it, you get error messages as you saw.

However, you are in luck, because by sheer coincidence, I just checked in (like 5 minutes ago), 2 working implementations of EEPROM: one for AVR processors, and one for ESP8266/ESP32 processors. Please grab the latest version from the `develop` branch, and read the instructions at:

* https://github.com/bxparks/EpoxyDuino/tree/develop/libraries/EpoxyPromAvr
* https://github.com/bxparks/EpoxyDuino/tree/develop/libraries/EpoxyPromEsp

Each directory has an `examples` directory, which I hope is enough to explain how to use these.

bxparks, 2021-02-25:
Closing for house cleaning. Pls reopen if you have updates.

closed 2021-02-25 by bxparks