bxparks/EpoxyDuino #34: snprintf formatting does not honour AVR '%S' (i.e. not '%s') for printing F() flash strings

brewmanz, 2021-08-04:
My sample failing test (that works on an Arduino AVR Nano) is
```
test(T1272_GoldStandard_Test_snprintf_9) {
  char buff128[128] = "?";
  auto str1 = "Resolved";
  auto str2 = F("Hello9");
  auto str3 = F("not this");
  str3 = F("World");
  snprintf(buff128, ARRAY_SIZE(buff128), "<%s><%S><%S>", str1, str2, str3);
  assertEqual("<Resolved><Hello9><World>", buff128);
}
```
Note that traditionally (whaatever that means), '%S' is used for UNICODE(wchar_t) strings, but AVR uses it for ASCII strings in Flash(Programme) memory.. 
When compiling my test, it gives the warning: 
```
TestGetFlashString.ino: In member function ‘virtual void test_T1272_GoldStandard_Test_snprintf_9::once()’:
TestGetFlashString.ino:34:42: warning: format ‘%S’ expects argument of type ‘wchar_t*’, but argument 5 has type ‘const __FlashStringHelper*’ [-Wformat=]
   34 |   snprintf(buff128, ARRAY_SIZE(buff128), "<%s><%S><%S>", str1, str2, str3);
      |                                          ^~~~~~~~~~~~~~        ~~~~
      |                                                                |
      |                                                                const __FlashStringHelper*
```

bxparks, 2021-08-06:
You are calling the `snprintf()` on the native host (Linux, MacOS, FreeBSD?) so of course it does not know about `%S`. (In fact, I didn't even know about `%S`, I cannot find any documentation after a few minutes of internet searching). Unfortunately, there is little EpoxyDuino can do to fix this. The only workaround I can suggest is to use some conditional code:

```C++
test(T1272_GoldStandard_Test_snprintf_9) {
  char buff128[128] = "?";
  auto str1 = "Resolved";
  auto str2 = F("Hello9");
  auto str3 = F("not this");
  str3 = F("World");
#if defined(EPOXY_DUINO)
  snprintf(buff128, ARRAY_SIZE(buff128), "<%s><%s><%s>", str1, str2, str3);
#else
  snprintf(buff128, ARRAY_SIZE(buff128), "<%s><%S><%S>", str1, str2, str3);
#endif
  assertEqual("<Resolved><Hello9><World>", buff128);
}
```

bxparks, 2021-08-18:
I don't see any reasonable way to fix this under POSIX environment. Closing.

closed 2021-08-18 by bxparks