bxparks/EpoxyDuino #38: IPAddress::toString() needed

hsaturn, 2021-09-18:
Hello Brian, could you please add this patch to IPAddress

I need this in order to enhance my tests around EspMock / TinyMqtt

Best regards

diff --git a/cores/epoxy/IPAddress.cpp b/cores/epoxy/IPAddress.cpp
index a7df382..3e835d6 100644
--- a/cores/epoxy/IPAddress.cpp
+++ b/cores/epoxy/IPAddress.cpp
@@ -114,3 +114,17 @@ size_t IPAddress::printTo(Print& p) const
     n += p.print(_address.bytes[3], DEC);
     return n;
 }
+
+String IPAddress::toString() const
+{
+    String s;
+
+    size_t n = 0;
+    for (int i =0; i < 3; i++)
+    {
+        s += String(_address.bytes[i]);
+        s += '.';
+    }
+    s += String(_address.bytes[3]);
+    return s;
+}
diff --git a/cores/epoxy/IPAddress.h b/cores/epoxy/IPAddress.h
index 2c0c1b6..5e1e504 100644
--- a/cores/epoxy/IPAddress.h
+++ b/cores/epoxy/IPAddress.h
@@ -67,6 +67,7 @@ public:
     IPAddress& operator=(uint32_t address);
 
     virtual size_t printTo(Print& p) const;
+    String toString() const;
 };
 
 const IPAddress INADDR_NONE(0,0,0,0);

bxparks, 2021-09-18:
As far as I can tell, `IPAddress::toString()` is defined only for ESP8266 and ESP32. So I guarded the method around a `#if defined(EPOXY_CORE_ESP8266)` so that it is available when `EPOXY_CORE := EPOXY_CORE_ESP8266` is defined in the `Makefile`.

Added some tests to validate.

If you need this on AVR, STM32, and other non-ESP platforms, you can use the `printTo(Print&)` function into the `PrintStr<N>` class from my [AceCommon library](https://github.com/bxparks/AceCommon/blob/develop/src/print_str/README.md). See the `IPAddressTest.ino` file for an example).

closed 2021-09-18 by bxparks

hsaturn, 2021-09-19:
Thanks a lot Brian