bxparks/EpoxyDuino #74: Simulated time

hsaturn, 2022-12-28:
Hello bxparks

My project (TinyMqtt) unit tests needs to simulate the time.
This helps to
- Debuggin: millis() does not increase making debug session based on time impossible
- Simplify unit testing
- Allow very precise time management for code that needs to deal with delays
- Allow to write unit tests that run quickly for unit tests that make use of delay()

The last point is important, because AUnit fails if unit tests take too long !

I've modified Arduino.* in order to achieve this.
All is working very well for TinyMqtt / EspMock with these modifications.
By the way, thank you for mentioning EspMock in your README, this mock now support network simulation !

Here are the modifications:

``` patch
diff --git a/cores/epoxy/Arduino.cpp b/cores/epoxy/Arduino.cpp
index 1f196da..387dbda 100644
--- a/cores/epoxy/Arduino.cpp
+++ b/cores/epoxy/Arduino.cpp
@@ -17,6 +17,9 @@
 #include <time.h> // clock_gettime()
 #include "Arduino.h"
 
+unsigned long epoxy_micros;
+bool epoxy_real_time = true;
+
 // -----------------------------------------------------------------------
 // Arduino methods emulated in Unix
 // -----------------------------------------------------------------------
@@ -36,25 +39,94 @@ int analogRead(uint8_t /*pin*/) { return 0; }
 void analogWrite(uint8_t /*pin*/, int /*val*/) {}
 
 unsigned long millis() {
-  struct timespec spec;
-  clock_gettime(CLOCK_MONOTONIC, &spec);
-  unsigned long ms = spec.tv_sec * 1000U + spec.tv_nsec / 1000000UL;
-  return ms;
+  if (epoxy_real_time)
+  {
+    struct timespec spec;
+    clock_gettime(CLOCK_MONOTONIC, &spec);
+    unsigned long ms = spec.tv_sec * 1000U + spec.tv_nsec / 1000000UL;
+    return ms;
+  }
+  else
+  {
+    return epoxy_micros / 1000;
+  }
+}
+
+void set_real_time()
+{
+  epoxy_real_time = true;
+}
+
+void set_seconds(unsigned long s)
+{
+  epoxy_micros = s * 1000 * 1000;
+  epoxy_real_time = false;
+}
+
+void add_seconds(unsigned long s)
+{
+  epoxy_micros += s * 1000 * 1000;
+  epoxy_real_time = false;
+}
+
+void set_millis(unsigned long ms)
+{
+  epoxy_micros = 1000 * ms;
+  epoxy_real_time = false;
+}
+
+void add_millis(unsigned long delta_ms)
+{
+  epoxy_micros += 1000 * delta_ms;
+  epoxy_real_time = false;
+}
+
+void set_micros(unsigned long us)
+{
+  epoxy_micros = us;
+  epoxy_real_time = false;
+}
+
+void add_micros(unsigned long delta_us)
+{
+  epoxy_micros += delta_us;
+  epoxy_real_time = false;
 }
 
 unsigned long micros() {
-  struct timespec spec;
-  clock_gettime(CLOCK_MONOTONIC, &spec);
-  unsigned long us = spec.tv_sec * 1000000UL + spec.tv_nsec / 1000U;
-  return us;
+  if (epoxy_real_time)
+  {
+    struct timespec spec;
+    clock_gettime(CLOCK_MONOTONIC, &spec);
+    unsigned long us = spec.tv_sec * 1000000UL + spec.tv_nsec / 1000U;
+    return us;
+  }
+  else
+  {
+    return 1000*epoxy_micros;
+  }
 }
 
 void delay(unsigned long ms) {
-  usleep(ms * 1000);
+  if (epoxy_real_time)
+  {
+    usleep(ms * 1000);
+  }
+  else
+  {
+    epoxy_micros += 1000 * ms;
+  }
 }
 
 void delayMicroseconds(unsigned int us) {
-  usleep(us);
+  if (epoxy_real_time)
+  {
+    usleep(us);
+  }
+  else
+  {
+    delay(1000*us);
+  }
 }
 
 unsigned long pulseIn(
diff --git a/cores/epoxy/Arduino.h b/cores/epoxy/Arduino.h
index 6efd1e7..ce2b402 100644
--- a/cores/epoxy/Arduino.h
+++ b/cores/epoxy/Arduino.h
@@ -223,8 +223,18 @@ void analogWrite(uint8_t pin, int val);
   void analogWriteRange(uint32_t range);
 #endif
 
+// Simulated time
+void set_millis(unsigned long);
+void add_millis(unsigned long);
+void set_micros(unsigned long);
+void add_micros(unsigned long);
+void add_seconds(unsigned long);
+void set_seconds(unsigned long);
+void set_real_time();
+
 unsigned long millis();
 unsigned long micros();
+
 void delay(unsigned long ms);
 void delayMicroseconds(unsigned int us);
 ```

bxparks, 2023-01-02:
I understand the convenience of being able to mock the low-level functions of the Arduino API. But EpoxyDuino was never intended to be a mocking library. It was supposed to emulate the Arduino API as closely as possible on a Unix/Posix system. The thinking was that if some code needs to be unit tested, then it should be designed such that it can be tested within the microcontroller itself, instead of having to rely on mocking functions that exist only in EpoxyDuino. I accepted the `digitalReadValue()` and `digitalWriteValue()` functions because they seems innocuous enough, but I'm not sure how I feel about adding a lot more of these mocking functions. I feel like if EpoxyDuino does support them, they should be part of a unified framework, instead of adding each one on an ad hoc basis, one by one. Let me think about this a bit more.

hsaturn, 2023-01-03:
Hi Brian

I understand this point of view. I do not really consider time simulation as a mock, but exactly in the 'unit test' spirit that EpoxyDuino was designed for. But yes, this is no more under the microcontroller itself. 

Since I use EpoxyDuino, I find this library to be very powerfull and incredibly valuable for debugging some tricky code on the PC with a decent debugger. In fact, I even never unit testing on the microcontroller (maybe I should...).
Simulated time is not in my point of view a mock, but a real feature allowing to control more precisely the time and thus increase the power of the emulation wanted by Epoxy.

In my point of view or less: in my use of your libs
- AUnit allows me to unit test on github, which is a incredible feature
- EpoxyDuino helps me to do that by implementing most functions.
- EspMock helps to mock the WiFi network and allows to run the test within githib

I've forked EpoxyDuino just to add the simulated time and be able to continue to use AUnit on github with the help of EpoxyDuino and EspMock. Of course it's up to you to decide, but honestly, I don't feel comfortable by forking just for that. (I"d prefer my fork to be temporary). I'd prefer to stick to your repo and delete mine.

Of course, I do agree that simulated time won't run on a microcontroller and thus, breaks the rule that unit tests could be ran on the microcontroller also.

I do not have a solution too. Best would be to have a overview on how EpoxyDuino is used. Maybe very few uses Epoxy to run unit tests on the microcontroller.

My guess is that because EpoxyDuino make it possible to run tests on the PC, it is so more comfortable that noone would think to run it on the microcontroller.

My typical use case : I've added many features to TinyMqtt + released without even testing them on the microcontroller. (which maybe not a so good habit :-D)

----

That said, I can agree with you. The choice is hard to make and it's up to you.

Maybe a solution would be replace functions by 'hooks'.

EpoxyDuino sticks to the law : emulate api as close at the Arduino's one.
ExtDuino 'based' on EpoxyDuino uses hooks to change some behaviours.

Something like this (very bad code, just look at the idea behind):
(also bad prototypes, we don't care...)

``` c++
// EpoxyDuino changes:
long epoxy_millis() {...}     // Default implementation

auto millis = epoxy_millis;   // Definition to millis function
```

millis is now a ptr on any function that can be overriden in ExtDuino
``` c++
// ExtDuino -> overrides some implementations

long simulated_millis();    // Able to work as millis or in simulated time
long init_extduino()
{
  millis = simulated_mills;
}
```

Not a good design (I'd prefer to group functions by theme such as Time, under a class such as EpoxyTime which would be polymorphic, and would allow to override functions.

This way, we would keep EpoxyDuino in its orientation: provides an emulation of the API that *is* compatible with the microcontroller. And that allows to write unit tests designed for both uController and PC.

And ExtDuino (or EpoxyTimeDuino or whatever) : provides a way to change some implementations. And thus breaks the rule of 'runs on microcontroller'.

Another solution that may enforce to group functions together could be
``` c++
#include <memory>

long now();
namespace EpoxyDuino {

  class TimeInterface
  {
    public:
      virtual long millis() = 0;
      virtual long micros() = 0;
      virtual void delay(long) = 0;
      virtual void delayMicroseconds(long) = 0;
  };

  class TimeImpl : public TimeInterface
  {
    public:
      long millis() override { return now(); }
      long micros() override { return now(); }
      void delay(long) override {}
      void delayMicroseconds(long) override {}
  };

  std::unique_ptr<EpoxyDuino::TimeInterface> time(new EpoxyDuino::TimeImpl);

}

long millis() { return EpoxyDuino::time->millis(); }
long micros() { return EpoxyDuino::time->micros(); }
void delay(long t) { EpoxyDuino::time->delay(t); }
void delayMicroseconds(long t) { EpoxyDuino::time->delay(t); }

namespace EpoxyDuinoSimulatedTime {
  class TimeImpl : public EpoxyDuino::TimeInterface
  {
    public:
      long millis() override { return time; }
      long micros() override { return time; }
      void delay(long) override {}
      void delayMicroseconds(long) override {}

    private:
      long time;
  };

  void init_ext_duino()
  {
    EpoxyDuino::time.reset(new EpoxyDuinoSimulatedTime::TimeImpl);
  }
}
```

To be frankly, I like the first solution because it is very simple to implement at a low cost, and I like the second solution because functions are grouped by theme. But ... I don't really like  for some unknown reasons boths.

As you I have to think more about that too.

hsaturn, 2023-01-03:
I've just thought about this, but millis() is intended to return the same value while it is called within interrupts. Only simulated time or thing near this could match this. Don't know what to do with this but I had to say.