bxparks/EpoxyDuino #75: std::make_unique

hsaturn, 2022-12-29:
Hello, EpoxyDuino does not support std::make_unique.

I've used this implementation that work like a charm. https://github.com/bstreiff/cppbits/blob/master/make_unique.h

Is it possible to add it to EpoxyDuino please ?

Best regards

bxparks, 2023-01-02:
It probably related to EpoxyDuino defaulting to `-std=gnu++11`. It looks like `std::make_unique` is a [C++14 feature](https://en.cppreference.com/w/cpp/memory/unique_ptr/make_unique).

* AVR uses `gnu++11`: https://github.com/arduino/ArduinoCore-avr/blob/master/platform.txt#L28
* ESP32 2.0.5 also uses `gnu++11`: https://github.com/espressif/arduino-esp32/blob/master/platform.txt
* Adafruit SAMD is ag `gnu++11`: https://github.com/adafruit/ArduinoCore-samd/blob/master/platform.txt#L42
* STM32 is at `gnu++14`: https://github.com/stm32duino/Arduino_Core_STM32/blob/main/platform.txt#L64
* ESP8266 3.0.2 seems to be at `gnu++17`: https://github.com/esp8266/Arduino/blob/master/platform.txt#L43
* Teeny3, Teensy4 seems to be at`gnu++14`:
    *  https://github.com/PaulStoffregen/cores/blob/master/teensy3/Makefile#L103
    * https://github.com/PaulStoffregen/cores/blob/master/teensy4/Makefile#L127
    * It's hard to tell because Teensy installs its Arduino core in a different way than other cores

I don't know. As you saw, I am getting tired of C++ and I don't have the energy to keep up with the exploding complexity of C++ anymore. But if ESP8266 uses gnu++17, and you want to compile ESP8266 code, I guess EpoxyDuino should support that. So the solution seems to be to use `gnu++17` when `EPOXY_CORE` is set to `EPOXY_CORE_ESP8266`.

I might be able to look at this in about 1-2 weeks...

hsaturn, 2023-01-03:
Hello Brian

I knew about the fact that C++ becomes boring for you (same for me in fact). And you're right, make_unique is not supported every where on arduino's ides so....

You may close this 'issue'. I'll deal with this another way.

best regards

bxparks, 2023-06-19:
So may be the long term solution is to add additional `EPOXY_CORE_XXX` flavors, so that different flavors specify different C++ versions of the different Arduino targets.

So something like:

* `EPOXY_CORE_ARDUINO_AVR`: gnu++11
* `EPOXY_CORE_SPARKFUN_AVR`: gnu++11
* `EPOXY_CORE_ATTINY`: gnu++11
* `EPOXY_CORE_STM32`: gnu++14
* `EPOXY_CORE_SEEEDUINO_SAMD`: gnu++14
* `EPOXY_CORE_ARDUINO_SAMD`: (not compatible with EpoxyDuino)
* `EPOXY_CORE_SPARKFUN_SAMD`: gnu++11
* `EPOXY_CORE_ADAFRUIT_SAMD`: gnu++11
* `EPOXY_CORE_ADAFRUIT_NRF52`: gnu++11
* `EPOXY_CORE_ESP8266`: gnu++17
* `EPOXY_CORE_ESP32`: gnu++11
* `EPOXY_CORE_TEENSYDUINO`: gnu++14

?? These are just the ones that I know about.

This is a lot of work, which is not very interesting, so it has not been high on my priority list. My personal solution is to just stick to C++11, so that I don't have to worry about compatibility with all the different Arduino APIs out there.

bxparks, 2023-06-19:
I guess since there is currently a `EPOXY_CORE_ESP8266` flavor, it is probably worth changing that target to gnu++17, to match the C++ version used by the actual ESP8266 Core.