bxparks/EpoxyDuino #77: Including LoRa library make error - use of undeclared identifier 'digitalPinToInterrupt'

timt, 2023-04-19:
When I include the LoRa library in the make file it produces the following errors:

/Users/xyz/Documents/Arduino/libraries/LoRa/src/LoRa.cpp:283:42: error: use of undeclared identifier 'B1000'
  if (readRegister(REG_FREQ_ERROR_MSB) & B1000) { // Sign bit is on
                                         ^
/Users/xyz/Documents/Arduino/libraries/LoRa/src/LoRa.cpp:369:24: error: use of undeclared identifier 'digitalPinToInterrupt'
    SPI.usingInterrupt(digitalPinToInterrupt(_dio0));

Anyone know how to resolve this?

bxparks, 2023-04-19:
The `Bxxxxxxxx` constants and `digitalPinToInterrupt()` aren't implemented in EpoxyDuino. It'd be awesome if you could send a PR to support the library that you are using. It's likely that there are other constants and functions missing. The code is here: https://github.com/bxparks/EpoxyDuino/tree/develop/cores/epoxy

(I'm traveling currently and will be slow to respond...)

bxparks, 2024-07-25:
I don't have the bandwidth to implement every single feature of the Arduino programming framework. I have to depend on the end-users to send me PRs for things that are missing and are important to them. Since this is over a year old, I'm closing this as stale.

bxparks, 2024-07-25:
(Very confusing GitHub UI wrt to "Closing"...)

closed 2024-07-25 by bxparks