bxparks/EpoxyDuino #80: sizeof(int) gives 4, while AVR int is 2 bytes long

cyborgdennett, 2023-05-07:
Hello, I was doing some programming and I found that EpoxyDuino returns 4 on sizeof(int). while in arduino it is said that an int is 2 bytes. 

I used the followingin the makefile.
```
EPOXY_CORE := EPOXY_CORE_AVR
```

cyborgdennett, 2023-05-07:
One way to fix it is by doing 
```
#ifdef EPOXY_DUINO
#define int int16_t
#endif
```
in your header file, but it will break the libraries you use. Any way to 'inject' it into other libraries?

bxparks, 2023-05-08:
EpoxyDuino is not meant to be a perfect simulator of an AVR microcontroller. It aims to support the Arduino API under a Unix environment. The `sizeof(int)` is not defined by the C/C++ language, but is implementation defined. This means that the sizeof(int) is the size given by the C++ compiler under Unix, which is 4, which happens to be the same in all Arduino environments except for AVR. I mention this potential issue with EpoxyDuino at the top of the README (https://github.com/bxparks/EpoxyDuino).

If you have code that assumes sizeof(int) == 2, it is non-portable code and will not work under EpoxyDuino. If you have control over that code, the solution is to change it explicitly to `int16_t`, so that it becomes portable. If you don't have control over that code, then you will need to fork it and fix it, and maybe upstream it to the original author of that library.

closed 2023-05-08 by bxparks