bxparks/EpoxyDuino #89: EpoxyDuino doesn't compile utility folder

curtisgalloway, 2024-07-15:
When trying to use the library [Adafruit_Motor_Shield_V2_Library](https://github.com/adafruit/Adafruit_Motor_Shield_V2_Library) with EpoxyDuino, it fails to compile because the Adafruit library includes some files in a subdirectory "utilitiy".  If I hack the library source in my Arduino libraries directory to put all the files in the top-level library directory, it works as expected.

The older Arduino library folder format allows for source in a "utility" directory; see the [Arduino library specification](https://arduino.github.io/arduino-cli/1.0/library-specification/#15-library-format-rev-22).  EpoxyDuino should handle those extra files.

curtisgalloway, 2024-07-15:
The new library format uses a 'src' subdirectory and has a property in the library.properties file to determine whether subdirectories of 'src' should be compiled recursively.

bxparks, 2024-07-24:
I'm actually surprised that you are able to compile these Adafruit libraries using EpoxyDuino. Because Adafruit libraries tend to have a lot of dependencies to other Adafruit libraries, which then depend on low-level hardware functions or symbols, which are not well supported by EpoxyDuino (since it's only a rough approximation of Arduino on a Unix-like environment).

I don't remember seeing the `utility` folder when reading the [library specification](https://arduino.github.io/arduino-cli/1.0/library-specification/#15-library-format-rev-22) when I created EpoxyDuino long ago, before the Arduino CLI existed. Maybe they updated it when they created the Arduino CLI. Maybe I just missed it.

**Research**

Reading the current version, it's clear that **source** files (`*.cpp`, `*.c`) files can go into `utility`. But it is not clear to me that header files (`*.h`) which are in `utility` are expected to be exported through the `-I` (--include) flag of the compiler. The [Adafruit_Motor_Shield_V2_Library](https://github.com/adafruit/Adafruit_Motor_Shield_V2_Library) library has examples of both assumptions:

a) The [Adafruit_MotorShield.h](https://github.com/adafruit/Adafruit_Motor_Shield_V2_Library/blob/master/Adafruit_MotorShield.h#L20) file has this line:

```c++
#include "utility/Adafruit_MS_PWMServoDriver.h"
```

which suggests that `utility` is **not** in the search path (otherwise the `utility/` prefix would not be necessary).

b) But [Adafruit_MotorShield.cpp](https://github.com/adafruit/Adafruit_Motor_Shield_V2_Library/blob/master/Adafruit_MotorShield.cpp#L33) contains:

```C++
#include <Adafruit_MS_PWMServoDriver.h> 
```
which is located at `utility/Adafruit_MS_PWMServoDriver.h`, which can only work if `utility` **is** in the header search path.

I guess (b) must take precedence over (a).

**Proposed Solution**

EpoxyDuino cannot selectively follow a v1.0 library format or a v1.5 format, depending on the existence of the `library.properties` file. (Maybe there is a way, but I don't know how to do it, and it will probably be a very complicated Makefile script.) What I can do is make EpoxyDuino support **both**, regardless of the existence of the `library.properties` file, with the assumption that a well-formed 3rd party library will follow either 1.0 or 1.5, but not both at the same time.

Can you test the following solution? It affects 3 lines of the `EpoxyDuino.mk` file at the root of the `EpoxyDuino` directory:

```
--- a/EpoxyDuino.mk
+++ b/EpoxyDuino.mk
@@ -155,7 +155,7 @@ CPPFLAGS += -I$(EPOXY_CORE_PATH)
 # Add the header files for libraries. Old Arduino libraries place the header
 # and source files right at the top. New Arduino libraries tend to use the
 # ./src/ subdirectory. We need to support both.
-CPPFLAGS_EXPANSION = -I$(module) -I$(module)/src
+CPPFLAGS_EXPANSION = -I$(module) -I$(module)/src -I$(module)/utility
 CPPFLAGS += $(foreach module,$(EPOXY_MODULES),$(CPPFLAGS_EXPANSION))
 
 # Linker settings (e.g. -lm).
@@ -176,11 +176,13 @@ EPOXY_SRCS := $(wildcard $(EPOXY_CORE_PATH)/*.cpp) \
 # Later Arduino libraries put the source files under the src/ directory. Also
 # support 3 levels of subdirectories. Support both C and C++ libraries files.
 MODULE_EXPANSION_CPP = $(wildcard $(module)/*.cpp) \
+       $(wildcard $(module)/utility/*.cpp) \
        $(wildcard $(module)/src/*.cpp) \
        $(wildcard $(module)/src/*/*.cpp) \
        $(wildcard $(module)/src/*/*/*.cpp) \
        $(wildcard $(module)/src/*/*/*/*.cpp)
 MODULE_EXPANSION_C = $(wildcard $(module)/*.c) \
+       $(wildcard $(module)/utility/*.c) \
        $(wildcard $(module)/src/*.c) \
        $(wildcard $(module)/src/*/*.c) \
        $(wildcard $(module)/src/*/*/*.c) \
```

curtisgalloway, 2024-07-25:
That works for compiling and running my project, thanks!  Now I can investigate what functionality I need to mock to test the “business logic” in my code of properly responding to motor encoding wheel events and dealing with limit switch input.

bxparks, 2024-07-25:
Cool, thanks for verifying

closed 2024-07-25 by bxparks

bxparks, 2024-07-25:
Released v1.6.0 with this fix.

commit eed664ce (2024-07-25): EpoxyDuino.mk: support v1.0 libraries which places files under the ./utility subdirectory (Fixes #89)